ngraziano/LMICPP-Arduino #3: ESP32 LoRa TTGO V1 board: TTN joining loops forever

Uewel, 2019-08-29:
Hi,
I've setup a new project in Platformio, added LMICPP as project library, use your ESP32_deepsleep example, set up the keys in TTN, put them into lorakeys.h, compile without errors and put everything onto my board. In platformio.ini I have added "build_flags = -DENABLE_SAVE_RESTORE"
After starting the firmware I've observed, that nothing will be received at TTN, even not after a long time. 

So, I have set -DLMIC_DEBUG_LEVEL=2 in platformio.ini and get the following output on the console:

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8
4217 Chip version : 0
4259 Init Default Channel
4272 Scheduled job 0x3ffbfd54, atRun 4271
4284 Running job 0x3ffbfd54, deadline 4271
4310 Init Join loop : avail=4267 txend=379449
4553 Scheduled job 0x3ffbfd70, atRun 4551
4785 Packet queued
4896 Running job 0x3ffbfd70, deadline 4551
5133 EV_JOINING
5224 Uplink join pending
5367 Airtime available at 379449 (previously determined)
5680 Uplink delayed until 379449
5864 Scheduled job 0x3ffbfd70, atRun 379324
379325 Running job 0x3ffbfd70, deadline 379324
379334 Uplink join pending
379341 Airtime available at 379449 (previously determined)
379374 Ready for uplink
379523 Time on air : 61 ms
379662 Setting  available time for band 1 to 764940
379949 Updating info for TX at 379340, airtime will be 3856.
380285 Updating global duty avail to 383196
380539 Config power to 16 on PA_BOOST
380753 TXMODE, freq=868100000, len=23, SF=7, BW=125, CR=4/5
381074 Scheduled job 0x3ffbfd70, atRun 381072
381329 Running job 0x3ffbfd70, deadline 381072
381589 Scheduled job 0x3ffbfd70, atRun 381587
381845 Running job 0x3ffbfd70, deadline 381587
382104 Scheduled job 0x3ffbfd70, atRun 382103
382361 Running job 0x3ffbfd70, deadline 382103
382619 Scheduled job 0x3ffbfd70, atRun 382618
382876 Running job 0x3ffbfd70, deadline 382618
383135 Scheduled job 0x3ffbfd70, atRun 383133
383391 Running job 0x3ffbfd70, deadline 383133
383650 Scheduled job 0x3ffbfd70, atRun 383649
383906 Running job 0x3ffbfd70, deadline 383649
384165 Scheduled job 0x3ffbfd70, atRun 384164
384422 Running job 0x3ffbfd70, deadline 384164
384681 Scheduled job 0x3ffbfd70, atRun 384680
384937 Running job 0x3ffbfd70, deadline 384680
385196 Scheduled job 0x3ffbfd70, atRun 385195
385453 Running job 0x3ffbfd70, deadline 385195
385712 Scheduled job 0x3ffbfd70, atRun 385710
385968 Running job 0x3ffbfd70, deadline 385710
386227 Scheduled job 0x3ffbfd70, atRun 386226
386483 Running job 0x3ffbfd70, deadline 386226
.....
(and so on) This never ends.

I'm using a TTGO ESP32 LoRa V1 board. Previous tests with generic LMIC library and it's ABP and OTAA examples was running with TTN. 
I have searched for a deep sleep aware library and found yours. Btw great work! My TTGO board should be similar to yours by Hetech. I2C settings are identical.

Do you have any idea from the log what goes wrong? 

Thank you very much in advance!

Regards,
Uwe

ngraziano, 2019-08-29:
In your log you have :
`Chip version : 0`
you should have 
`Chip version : 18`

It indicate the ESP can not communicate with the radio chip.

Can you check you have the right pin number for `.nss` and you have `SPI.begin();` at the beginning of `Setup` and check again if you avec Chip version : 18.

I add some logs to the library and I correct a buffer size in the sample.
Make sure you remove `.pio` to force plaformio to reload the library.

Uewel, 2019-08-29:
Thank you for your fast answer.
In the meantime I have changed SPI.begin() into SPI.begin(5,19,27,18). This was also necessary for me with the generic LMIC lib. With that the example code seems to run. It does the joining, sends the temperature value and goes to deep sleep. I have verified this with a ampere meter. Then after some time (I assume after the configured 135s) it wakes up again (higher current) and brings some debugging info and goes to deep sleep again. 
But from now it doesn't send further data to TTN. Mabye this is the normal behavior of your example code. I will retest with your changes (and understand the example code).

I'm so sorry, that I was so unpatient with creating this issue.

ngraziano, 2019-08-29:
I am currently testing the code, and at some point I break the save state and restore state. 
I am still investigating.

Uewel, 2019-08-29:
Ok. I will wait for your response before I continue my testing tomorrow morning. Thank in advance.

ngraziano, 2019-08-29:
It should be good now.
Make sure you update the library by deleting `.pio`. And check you have the last main.cpp of the sample (i mess a little saveState variable).

The example should join and send a value then sleep for 135s (TX_INTERVAL) and wakeup and send a value (every 135s).

Uewel, 2019-08-30:
It is really good now. It works perfect as expected. Thank you very much!

I have modified my TTGO board by removing the onboard LDO for 3.3V. With that I can supply 3.3V from external and measure it's current. With your example I measure around 880uA. This current I have also measured with a bare deep-sleep example code, when I don't disable RTC memory (hibernate). But of course, this memory will be needed for LoRa state saving now. The TTG board (V1) is not very optimal for power saving. But 0,8mA during deep-sleep is quite ok.

I will now add my BME280 code and further the code for a ePaper display.

Again, thank you very much for your support as well as for this project!

closed 2019-08-30 by ngraziano