ngraziano/LMICPP-Arduino #5: Heltec wireless stick: TTN Sees the device's packet, but device is stuck in RX timeout

ioangogo, 2020-01-15:
Hi, im currently having an issue where the device is being seen by the TTN network and is showing up as seen, but the logs constantly show this

```
66216 Chip version : 252
66274 Init Join loop : avail=66269 txend=4294777017
EventType::JOINING
66292 Ready for uplink
66319 Time on air : 61 ms
67668 Config power to 16 on PA_BOOST
70142 TXMODE, freq=868100000, len=23, SF=7, BW=125, CR=4/5
74037 Not using interupt trigger 0
76382 End TX  74036
77749 Rx delay : 4873 ms
378654 RXMODE_SINGLE, freq=868100000, SF=7, BW=125, CR=4/5, IH=2
394995 Not using interupt trigger 0
395002 RX timeout
395008 End RX - Start RX : 261792 us 
395016 Rx delay : 5885 ms
441842 RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=2
472583 Not using interupt trigger 0
472589 RX timeout
472595 End RX - Start RX : 491424 us 
472603 Next available : 472602 , Choosen 643078
473602 Uplink delayed until 643078
642956 Ready for uplink
642965 Time on air : 61 ms
642979 Config power to 16 on PA_BOOST
642996 TXMODE, freq=868300000, len=23, SF=7, BW=125, CR=4/5
646865 Not using interupt trigger 0
646967 End TX  646863
648462 Rx delay : 4873 ms
951468 RXMODE_SINGLE, freq=868300000, SF=7, BW=125, CR=4/5, IH=23
967811 Not using interupt trigger 0
967816 RX timeout
967823 End RX - Start RX : 261616 us 
967829 Rx delay : 5885 ms
1014717 RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=23
1045456 Not using interupt trigger 0
1045462 RX timeout
```

Here is the pin map
```C
const lmic_pinmap lmic_pins = {
    .nss = 18,
    .prepare_antenna_tx = nullptr,
    .rst = 14,
    .dio = {DIO0, DIO1}
};
```

and here is the code i am using to init:

```C
void ttnHandling(void * param){
    SPI.begin(SCK,MISO,MOSI,lmic_pins.nss);
    os_init();
    LMIC.init();
    LMIC.reset();
    LMIC.setEventCallBack(onEvent);
    SetupLmicKey<appEui, devEui, appKey>::setup(LMIC);
    LMIC.setClockError(MAX_CLOCK_ERROR * 3 / 100);
    //LMIC.setAntennaPowerAdjustment(-14);

    LMIC.startJoining();

    while(true){
        OsDeltaTime to_wait = OSS.runloopOnce();
        int towait = to_wait.to_ms();
        if (to_wait > OsDeltaTime(0)){
            vTaskDelay(portTICK_PERIOD_MS*towait);
        }
        
        //Give other tasks a chance to run on the processor
    }
    
}
```

ioangogo, 2020-01-15:
Testing with the example seems to work, it might be an issue with me making use of the multi-tasking capabilities of the device, close if you want to

ngraziano, 2020-01-15:
Hi,

The timing to detect the end of TX and begin RX is critical. (gateway send data 5s after end of end of TX). 
I never try to use vTaskDelay.  For test can you try without it just to see if this is the problem.

You can also try to put something like 
```cpp
       const delayfor4ms = xxxx;
       if (to_wait > OsDeltaTime::from_ms(5){
            vTaskDelay(delayfor4ms);
        }
```

I will retry "examples/esp32" tomorrow to check if there is a problem.

ngraziano, 2020-01-18:
Hi,

I check again the esp32 sample and it is working. It miss the first join but the second one work well.
(I only remove LMIC.setAntennaPowerAdjustment(-14); and set the keys)

[image]

The corresponding log to compare to yours 
```
2624 Chip version : 18
2698 Init Join loop : avail=2678 txend=4294630925
2711 Packet queued
2724 Ready for uplink
2753 Time on air : 61 ms
2774 Config power to 16 on PA_BOOST
2964 TXMODE, freq=868300000, len=23, SF=7, BW=125, CR=4/5
6837 End TX  0
6848 Rx delay : 4873 ms
304585 RXMODE_SINGLE, freq=868300000, SF=7, BW=125, CR=4/5, IH=0
316076 Not using interupt trigger 0
316085 RX timeout
316096 End RX - Start RX : 183664 us
316107 Rx delay : 5885 ms
367835 RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
398578 Not using interupt trigger 0
398586 RX timeout
398596 End RX - Start RX : 491920 us
398608 Next available : 398607 , Choosen 853282
398673 Uplink delayed until 853282
853160 Ready for uplink
853171 Time on air : 61 ms
853188 Config power to 16 on PA_BOOST
853208 TXMODE, freq=868500000, len=23, SF=7, BW=125, CR=4/5
857078 Not using interupt trigger 0
857088 End TX  857077
857095 Rx delay : 4873 ms
1161649 RXMODE_SINGLE, freq=868500000, SF=7, BW=125, CR=4/5, IH=0
1174110 Not using interupt trigger 0
1174128 End RX - Start RX : 198976 us
1174151 Ready for uplink
1174167 Time on air : 46 ms
1174183 Config power to 16 on PA_BOOST
1174368 TXMODE, freq=867100000, len=14, SF=7, BW=125, CR=4/5
1177278 Not using interupt trigger 0
1177288 End TX  1177277
1177296 Rx delay : 974 ms
1238210 RXMODE_SINGLE, freq=867100000, SF=7, BW=125, CR=4/5, IH=0
1241945 Not using interupt trigger 0
1241953 RX timeout
1241964 End RX - Start RX : 60272 us
1241975 Decode Frame RX1
1241981 No downlink data
1242063 Rx delay : 1952 ms
1299335 RXMODE_SINGLE, freq=869525000, SF=9, BW=125, CR=4/5, IH=0
1307292 Not using interupt trigger 0
1307300 RX timeout
1307311 End RX - Start RX : 127328 us
1307322 Decode Frame RX2
1307327 No downlink data
9744927 Ready for uplink
9744939 Time on air : 46 ms
9744956 Config power to 16 on PA_BOOST
9744976 TXMODE, freq=868100000, len=14, SF=7, BW=125, CR=4/5
9745089 Packet queued
9747884 Not using interupt trigger 0
9747894 End TX  9747883
9747903 Rx delay : 974 ms
9808773 RXMODE_SINGLE, freq=868100000, SF=7, BW=125, CR=4/5, IH=0
9812507 Not using interupt trigger 0
9812515 RX timeout
9812526 End RX - Start RX : 59568 us
9812535 Decode Frame RX1
9812541 No downlink data
9812624 Rx delay : 1952 ms
9869961 RXMODE_SINGLE, freq=869525000, SF=9, BW=125, CR=4/5, IH=0
9877918 Not using interupt trigger 0
9877926 RX timeout
9877936 End RX - Start RX : 127648 us
9877947 Decode Frame RX2
9877952 No downlink data
```

After checking again your logs I see  `Chip version : 252` which is not normal, it should be `Chip version : 18` for an SX1276.
Can you check the nss pin number and that there is not other chip active on the SPI bus ?

ioangogo, 2020-01-18:
Just had an opportunity to get closer to a gateway, the timing fix seems to have worked.

Just range issues now that im having with the device now, but thats probably because of the terrible antenna that came with the device

ioangogo, 2020-01-26:
I do have 1 question, `LMIC.setAntennaPowerAdjustment(-14);` What effect can setting this have and is there a way to calculate an appropriate Adjustment based on the antenna?

ioangogo, 2020-01-26:
> After checking again your logs I see Chip version : 252 which is not normal, it should be Chip version : 18 for an SX1276.

ioangogo, 2020-01-26:
I will add that the chosen pins for this board are a bit odd 
[image]

ioangogo, 2020-01-26:
although looking at the code now, its a bit odd that im not tripping on that `ASSERT(v == 0x12);` possibly be a printing error?

ioangogo, 2020-01-26:
Yep, did some testing and edited that section to this

```C++
uint8_t const v = hal.read_reg(RegVersion);
uint8_t testv = hal.read_reg(RegVersion);
PRINT_DEBUG(1, F("Chip version : %u"), v);
Serial.printf("Var testv: Hex: 0x%02X Unsigned Int: %u\n", testv, testv);
```
it then prints this
```
66277 Chip version : 252
Var testv: Hex: 0x12 Unsigned Int: 18
```
So im assuming some compile time optimisation is setting the value of the const when it is printed, so the chip is returning the right value, but printing is printing the wrong thing

ioangogo, 2020-01-26:
Some reason reading twice gets the correct value

ioangogo, 2020-01-27:
I found why the chip number was wrong, one of the connected devices i was using was causing interferance :P

closed 2020-01-27 by ioangogo

ngraziano, 2020-01-27:
Hi, 
It's great you find the problem with the version number.

For the `LMIC.setAntennaPowerAdjustment(-14);`, it add the value to the calculate power.

The system begin with TX power at 14dB (the limit for EU868) and the ADR can reduce it to limit the power use. With setAntennaPowerAdjustment you can add value to the calculate power if you know you antenna gain.  (I don't know how to calculate the good value ). With this library you can TX with power from 2dB to 17dB (other value -4 to 2 and 17 to 20 are not handled)

But in real, I only use it to limit power consumption of the SX1276 chip because I have a bad 3.3v regulator on some boards and my devices are near my gateway.