ngraziano/LMICPP-Arduino #6: ESP32 + deepsleep + TTN ABP

chron0, 2020-01-25:
Ahoy, thanks for sharing this lib, I'm testing with OTAA now and that seems to work but I can't find an example how to do ABP with esp32 and deepsleep. The only references to ABP point to the old examples from the other lmic lib which gave me trouble with deepsleep and unsuccessful radio restarts on a ttgo t-beam with the axp192 power controller. I'd love to use your lib with ABP but I'm not sure how to implement it. If you ever find the time to add an ABP example, it would be very much appreciated. Thank you very much.

ngraziano, 2020-01-25:
Hi,
I only use OTAA with my library and never found the need to use ABP so I never try to make an example for this.  

I just check and there is missing public function to configure the needed information.
With APB you need to configure key with "setSession"
```cpp
  // ABP Set session information
  const uint32_t TTN_NET_ID = 0x000013;
  const uint32_t DEV_ADRESS = 0x26012940;
  const uint8_t NET_KEY[] = {0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
                             0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};
  const uint8_t APP_KEY[] = {0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
                             0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};

  AesKey appkey;
  std::copy(APP_KEY, APP_KEY + 16, appkey.begin());
  AesKey netkey;
  std::copy(NET_KEY, NET_KEY + 16, netkey.begin());
  LMIC.setSession(TTN_NET_ID, DEV_ADRESS, netkey, appkey);
```

but you will also need to configure RX2 datarate to DR3 (for EU868) but it is not possible for now with my lib and you need to configure all the channels for EU868 which is also not possible with my lib.

I will need to change some private methods to public, I will try to have a look to during the week if is easy to make this modification.

chron0, 2020-01-25:
I was assuming about as much from some comments in the code, where mostly the following part is done in the old examples to set this config up, I reckon:

```
    LMIC_setSession (0x1, DEVADDR, NWKSKEY, APPSKEY);
 
    LMIC_setupChannel(0, 868100000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);      // g-band
    LMIC_setupChannel(1, 868300000, DR_RANGE_MAP(DR_SF12, DR_SF7B), BAND_CENTI);      // g-band
    LMIC_setupChannel(2, 868500000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);      // g-band
    LMIC_setupChannel(3, 867100000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);      // g-band
    LMIC_setupChannel(4, 867300000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);      // g-band
    LMIC_setupChannel(5, 867500000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);      // g-band
    LMIC_setupChannel(6, 867700000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);      // g-band
    LMIC_setupChannel(7, 867900000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);      // g-band
    LMIC_setupChannel(8, 868800000, DR_RANGE_MAP(DR_FSK,  DR_FSK),  BAND_MILLI);      // g2-band

    // Disable link check validation
    LMIC_setLinkCheckMode(0);

    // TTN uses SF9 for its RX2 window.
    LMIC.dn2Dr = DR_SF9;

    // Set data rate and transmit power for uplink (note: txpow seems to be ignored by the library)
    LMIC_setDrTxpow(DR_SF7,14);
```

My use case is a mobile tracking beacon that should live as long as possible on battery (hence deep-sleep requirement) and reliably broadcast location data to TTN without having to establish a link first.

ngraziano, 2020-02-01:
I just add  a example for abp and deepsleep please take have a look and tell me if it is working for you.

chron0, 2020-02-02:
Great! Thanks for the info and ofc the actual updates to the code. I had already seen the commits and will try a fresh experiment, just with the example abp code to give you better feedback about the results on my platform.

chron0, 2020-02-02:
So far, so good, it seems to survive deepsleep resets - I haven't done any other power saving optimizations yet to keep things clear and easy:

```
rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
834769391 AXP192 Communication [Check]
834771315 Chip version : 252
834771372 Ready for uplink
834771411 Time on air : 1155 ms
834771432 Config power to 16 on PA_BOOST
834771452 TXMODE, freq=868300000, len=14, SF=12, BW=125, CR=4/5
834771709 Packet queued
834843658 Not using interupt trigger 0
834843668 End TX  834843657
834843678 Rx delay : 1049 ms
834843727 Charging: TRUE
834846947 getVbusVoltage: 5030.300293
834847063 getVbusCurrent: 427.125000
834850261 getBattVoltage: 3927.000000
834853449 getBattChargeCurrent: 386.500000
834853534 getBattDischargeCurrent: 0.000000
834856759 getTemp: 199.900009
834856841 getTSTemp: 800.000000
834950900 RXMODE_SINGLE, freq=868300000, SF=12, BW=125, CR=4/5, IH=0
834961163 Not using interupt trigger 0
834961172 RX timeout
834961181 End RX - Start RX : 830912 us 
834961190 Decode Frame RX1
834961198 No downlink data
834961342 Rx delay : 1991 ms
834961541 Charging: TRUE
834964820 getVbusVoltage: 5035.399902
834964907 getVbusCurrent: 427.125000
834968135 getBattVoltage: 3931.400146
834971323 getBattChargeCurrent: 419.000000
834971408 getBattDischargeCurrent: 0.000000
834974633 getTemp: 199.900009
834974716 getTSTemp: 800.000000
835039336 RXMODE_SINGLE, freq=869525000, SF=9, BW=125, CR=4/5, IH=0
835042430 Not using interupt trigger 0
835042438 RX timeout
835042448 End RX - Start RX : 1188544 us 
835042458 Decode Frame RX2
835042464 No downlink data
835042615 EV_TXCOMPLETE (includes waiting for RX windows)
835042943 State save len = 189
```
I can only tell tomorrow if TTN will actually receive it as I'm out of available repeater range and don't have a local one running yet :(

A couple of questions remain so far:

1. Why are the first the channels in the example list commented out?

```
  // Channel 0,1,2 : default channel for EU868
  // LMIC.setupChannel(0, 868100000, dr_range_map(0, 5));
  // LMIC.setupChannel(1, 868300000, dr_range_map(0, 5));
  // LMIC.setupChannel(2, 868500000, dr_range_map(0, 5));
  LMIC.setupChannel(3, 867100000, dr_range_map(0, 5));
  LMIC.setupChannel(4, 867300000, dr_range_map(0, 5));
  LMIC.setupChannel(5, 867500000, dr_range_map(0, 5));
  LMIC.setupChannel(6, 867700000, dr_range_map(0, 5));
  LMIC.setupChannel(7, 867900000, dr_range_map(0, 5));
```
2. What should be set to tell the radio to use max rf output power?

```  
// reduce power
// LMIC.setAntennaPowerAdjustment(-14);
```

3. What consequences has setting the first parameter of setTxData2
```
LMIC.setTxData2(3, &val, 1, false);
```

4. Regarding State: Is there a saved counter within the rtc saved state managed by the library which I haven't found yet? AFAIK each TX needs to have a increasing txid which I needed to manually track with other libs in ABP mode.

ngraziano, 2020-02-03:
> 1. Why are the first the channels in the example list commented out?

The first tree channels are defined in the library (the lorawan specification say they are mandatory), no need to redefine in user code, it is just as a reminder.

> 2. What should be set to tell the radio to use max rf output power?

By default the library set the output power to the maximum allowed for EU868.
You may need to set `LMIC.setLinkCheckMode(false);` to disable ADR otherwise the network may ask the device to decrease the power and change the datarate.

> 3. What consequences has setting the first parameter of setTxData2
The first parameter is the port, you can use 1 to 223 to differentiate your type of message and you will receive it on the application side.

> 4. Regarding State: Is there a saved counter within the rtc saved state managed by the library which I haven't found yet? AFAIK each TX needs to have a increasing txid which I needed to manually track with other libs in ABP mode.

I'm not sure to understand the question. The tx counter is saved in the RTC saved buffer by the command `LMIC.saveState(store);` it is done after each transmission (and other important values are also saved at same time).

chron0, 2020-02-03:
First test run today on a location where I usually get up to 3 gateways was not successful - but weather was really bad (so no live usb debugging out there) and the rain may have been a problem.

Just to be sure, Network Key and App Key are supposed to be MSB, right?

chron0, 2020-02-03:
> > 1. Why are the first the channels in the example list commented out?
> 
> The first tree channels are defined in the library (the lorawan specification say they are mandatory), no need to redefine in user code, it is just as a reminder.

Check!

> > 1. What should be set to tell the radio to use max rf output power?
> 
> By default the library set the output power to the maximum allowed for EU868.
> You may need to set `LMIC.setLinkCheckMode(false);` to disable ADR otherwise the network may ask the device to decrease the power and change the datarate.

Check. In other implementations I've seen this being done in the event handler cases like so:
```
   case EV_JOINED:
      LoraStatus = "JOINED";
      LMIC_setLinkCheckMode(0);
      break;
```
Should I do this here too or just in the setup phase as well?

> > 1. What consequences has setting the first parameter of setTxData2
> 
> The first parameter is the port, you can use 1 to 223 to differentiate your type of message and you will receive it on the application side.

ohh, ic. thx for clearing that up for me.

> > 1. Regarding State: Is there a saved counter within the rtc saved state managed by the library which I haven't found yet? AFAIK each TX needs to have a increasing txid which I needed to manually track with other libs in ABP mode.
> 
> I'm not sure to understand the question. The tx counter is saved in the RTC saved buffer by the command `LMIC.saveState(store);` it is done after each transmission (and other important values are also saved at same time).

I was talking about the "frame counter" - txid may be bad wording here but I think we're talking about the same.

ngraziano, 2020-02-04:
For `LMIC.setLinkCheckMode(false);` you should put it in Setup.
(In OTAA it is setup in handler because it is reset to true when device join)

ngraziano, 2020-02-04:
I just add a log message to help track of frame counter (frames up)
"Build pkt # 123456"

chron0, 2020-02-05:
I've updated my test setup and the log looks very promising:

```
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
AXP GetTempInAXP192: 26.20
AXP GetCoulombData: 157.65
AXP GetBatVoltage: 4.06
AXP GetBatCurrent: 0.00
AXP GetBatPower: 0.00
AXP GetBatChargeCurrent: 0.00
Charge-Current setting : 700.00 mA
AXP getSettingChargeCurrent: 700.00
AXP GetVinVoltage: 0.00
AXP GetVinCurrent: 0.00
AXP GetVBusVoltage: 5.19
AXP GetVBusCurrent: 55.12
get result:ff
AXP getLDO2Voltage: 3300
get result:ff
AXP getLDO3Voltage: 3300
AXP getDCDC1Voltage: 3300
SRC REG:0x84
Charging enable is enable
Charging target-voltage : 0x0
end when the charge current is lower than 10% of the set value
Charge current : 700.00 mA
292470 Chip version : 18
292576 Ready for uplink
292626 Build pkt # 0
292643 Time on air : 164 ms
292692 Config power to 2 on PA_BOOST
292736 TXMODE, freq=868300000, len=14, SF=9, BW=125, CR=4/5
292924 Packet queued
303054 Not using interupt trigger 0
303077 End TX  303051
303094 Rx delay : 1000 ms
365572 RXMODE_SINGLE, freq=868300000, SF=9, BW=125, CR=4/5, IH=0
367647 Not using interupt trigger 0
367664 RX timeout
367684 End RX - Start RX : 33472 us 
367703 Decode Frame RX1
367717 No downlink data
367744 Rx delay : 1991 ms
427575 RXMODE_SINGLE, freq=869525000, SF=9, BW=125, CR=4/5, IH=0
430673 Not using interupt trigger 0
430689 RX timeout
430709 End RX - Start RX : 50096 us 
430732 Decode Frame RX2
430745 No downlink data
430773 EV_TXCOMPLETE (includes waiting for RX windows)
431087 State save len = 189
431232 Shutting down DCDC/L
DO Rails on AXP
431887 Goodnight, going to sleep now...
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
[... 167 lines cut ...]
24293028 End RX - Start RX : 49872 us 
24293051 Decode Frame RX2
24293065 No downlink data
24293146 EV_TXCOMPLETE (includes waiting for RX windows)
24293469 State save len = 189
24293626 Shut
ting down DCDC/LDO Rails on AXP
24294263 Goodnight, going to sleep now...
```

I've noticed another issue you may like to hear about:

On clean reset, the chip version is always 18 in my case. Using your example code and putting the ESP into deepsleep with 

```
ESP.deepSleep(TX_INTERVAL.to_us());
```
works, an on subsequent deepsleep_restarts, the chip version remains being 18 (as I think it should).

Now, when putting the ESP to sleep like this
```
    esp_sleep_enable_timer_wakeup(wakeup_in_us);
    esp_deep_sleep_start();
```
as documented in many esp32 deep_sleep resources out on the net, then it doesn't work anymore. When the ESP wakes up from deepsleep, the chip version is either 0 or 252 but never 18. Not sure what to make of it but if someone else is having that problem, they may want to change the way they put the ESP into deepsleep. Perhaps esp_deep_sleep_start() is a more esspressif function and ESP.deepSleep() is an arduino wrapper?

chron0, 2020-02-06:
Still no luck with a test run today :/ 

[image]

I'll be back in the office next week, so I can test and watch it properly at the same time and I suppose I'll have to get a gateway myself as well to be able to debug on the receiving side...

ngraziano, 2020-02-07:
Without an gateway the diagnostic is more complicated. 
For me it works.

Network session key and app session key are msb ( I only need to click on '<>' button before copy to code to NET_KEY and APP_KEY)

mikethebee, 2020-02-15:
@ngraziano Hello and thanks for your hard work on this lib. I have a similar need to deep-sleep for battery use. I maybe missing something with the removal of "rejoin" from otaa, but does this not cause a new join after waking? ABP should avoid starting a new join session each time, so I am looking this discussion with interest and also trying to NVM session storing for otaa rejoin. btw @chron0 there is a TTN indoor gateway available for under 100euro. thx, Mike

ngraziano, 2020-02-15:
Hi @mikethebee,

The REJOIN part of LMIC library is not in 1.0.x standard of lorawan (it is a feature of 1.1x standard and if I understand and it is only  for roaming between networks) it is why I remove it. It used to send a JOIN request when the device haven't receive data for a long time to try to join another network.
 
Concerning OTAA and deepsleep : the library is saving the session state so after wake up there is no need to join again. If you try the example `esp32-deepsleep` you will see only one JOIN at startup. After the session state is saved in a buffer in RTC RAM which survive deep sleep (if you want it elsewhere you can override StoringAbtract  and RetrieveAbtract ). There is no difference for the library if the system sleep or not. 
The only critical point is that you need to send at least every one or hours because OsTime overflow and band usage may induce problems.

ABP also need to save some data in deep sleep, because the information between send is not only the network session key, application session key and device id. It need to save the current ADR state, the usage of bands... In fact now it save the same data than OTAA version.

I am not a big fan of ABP for my devices, I only use OTAA and it is working even with deep sleep.

If you have more question on deepsleep and OTAA maybe open another bug. 
Regard,
Nicolas

closed 2020-04-30 by ngraziano