ngraziano/LMICPP-Arduino #8: DIO2 as a RF switch is working ?

marciolm, 2020-02-09:
Hello. 
I am trying to use an Sx1262 module  (Ebyte E22900M20S) + ESP32. There is an external circuit to control the RF switch using the module's external pin DIO2, but this pin never changes its state when I run the examples. The gateway receives a very weak signal of about -108 dBm. If I force the module's pin TXEN to high, I got -38 dBm. 

The only changes in "esp32" example were:
`constexpr lmic_pinmap lmic_pins = {
    .nss = 5,
    .prepare_antenna_tx = nullptr,
    .rst = 4,
    .dio = {/* busy */ 22, /* DIO1 */ 21},
};
RadioSx1262 radio{lmic_pins, ImageCalibrationBand::band_902_928 };
LmicUs915 LMIC{radio, OSS};
`
And I tried to insert the `RadioSx1262 set_DIO2_as_rf_switch_ctrl();` in the setup() session, with no success. 
Can you confirm the DIO2 is actually controlling the RF switch? Alternatively, is there a way to use a pair of microcontroller GPIOs to control the TXEN and RXEN, with no GPIO2?
Thanks
Márcio

ngraziano, 2020-02-09:
Hello,

For my board I do not need to set DIO2 to control TX so I do not call set_DIO2_as_rf_switch_ctrl().

You can try following
- call `radio.set_DIO2_as_rf_switch_ctrl();` in setup but after `LMIC.init()` (LMIC.init reset the chip)
- or make a copy of the library in you lib folder of the project and edit file radio_sx1262.cpp at line 555 to uncomment `// set_DIO2_as_rf_switch_ctrl();` 
- or in lmic_pins  define an lambda function in prepare_antenna_tx   to set the pin you need to control TXEN and RXEN
```cpp
constexpr lmic_pinmap lmic_pins = { 
    .nss = 5,
    .prepare_antenna_tx = [](bool isTx) {
        if (isTx)
        {

            digitalWrite(PIN_FOR_RXEN, LOW);
            digitalWrite(PIN_FOR_TXEN, HIGH);
        }
        else
        {
            digitalWrite(PIN_FOR_TXEN, LOW);
            digitalWrite(PIN_FOR_RXEN, HIGH);
        }
    },
    .rst = 4,
    .dio = {/* busy */ 22, /* DIO1 */ 21} };
```

I hope it will help.

Nicolas

marciolm, 2020-02-10:
Hello Nicolas
The first option did not work, but the second one worked!
Thank you!
Márcio

closed 2020-02-10 by marciolm