ngraziano/LMICPP-Arduino #16: Will this code be compatible with Arduino ide?

Muuhoooo, 2021-11-24:
This code and repository can use only vs. code but I want work  Arduino ide with SX1262 E22_900M22S module and ESP32 WROWER_B module. Actually, I'm just trying to send data to Helium console with OTAA. Will this code be compatible with Arduino ide  and can ı send data over OTAA?

ngraziano, 2021-11-24:
This code may be usable with Arduino IDE for ESP32 (with AVR it is not because of stdlib) but I have never try and I don't have interest and  time to test. I recommend to use Platformio 

I have never try to use it on Helium network (only TTN), can you tell me if you manage to make it work?

Muuhoooo, 2021-11-24:
Thank you for your return. I am looking for arduino compatible library directly. Experience this code if I can't find it, I'll come back here.

closed 2021-11-24 by ngraziano