ngraziano/LMICPP-Arduino #32: Why only sx1262 with TXCO?

Rustie0125, 2023-04-18:
Hello, just wanted to findout the reason for only supporting sx1262 with tcxo. ?

ngraziano, 2023-04-18:
Just because I only have module using sx1262 with tcxo. I can't test without the hardware.

Rustie0125, 2023-04-19:
Lol the only hardware I have is probably 20 sx1262 without tcxo only :-).
Really want to use the 328P with lorawan but can't find any library that
will work

Rustie0125, 2023-04-19:
Is there any reason it wont work ? or you just have not tested it yet ?

Rustie0125, 2023-04-19:
im based in Canada using the 915 band I changed the band under the simple 1262 example as follows 
```
// Radio class for SX1262
RadioSx1262 radio{lmic_pins, ImageCalibrationBand::band_902_928};
// Create an LMIC object with the right band
LmicEu868 LMIC{radio};
OsTime nextSend;
```

But get this output on the serial monitor, two questions any other place I need to change bands ? I also need to see where to change the sub and from sub 2 to sub 3 if possible, not sure if i should just change the default freq 

Thanx for the help

```
49353 Time before send 134s
51164 Sleep 0s
52148 Time before send 134s
53959 Sleep 0s
54943 Time before send 134s
56754 Sleep 0s
57739 Time before send 134s
59549 Sleep 0s
60533 Time before send 134s
62344 Sleep 0s
63328 Time before send 134s
65139 Sleep 0s
66123 Time before send 134s
67934 Sleep 0s
72470 Ready for uplink
72804 Time on air : 61 ms
72866 Init Configure
74516 Status 32 mode : 3, status 1
75729 TXMODE, freq=868500000, len=23, SF=7, BW=125, CR=4/5
```

Rustie0125, 2023-04-19:
Cool, Got it working on Sx1262 without TCXO crystal so you can remove that disclaimer. it take a fair amount of time before it joins via OTA but it does work. Still need to see how to change sub bands, sf ect cuase its not clear but making progress! out of interest did you use a low power library to go into sleep or did you never to sleep current optimization ?

ngraziano, 2023-04-22:
Hi sorry for the delay of response. I only look at this project on my free time.

The limitation for TXCO is due to the init phase and the use of DIO3 for controlling TXCO (https://github.com/ngraziano/LMICPP-Arduino/blob/master/src/lmic/radio_sx1262.cpp#L577). If there is nothing if there is nothing connect on this pin there is no problem.

If you are in canada you should use LmicUs915 instead of LmicEu868 to have the right band.

For the the low power with 328P you can look at https://github.com/ngraziano/LMICPP-Arduino/tree/master/examples/balise there is a simple exemple.

Please use discussion tab to ask question next time and keep Issues tab for real bug.

closed 2023-04-30 by ngraziano