ngraziano/LMICPP-Arduino #42: arduino standalone version?

areytechai, 2024-09-29:
Is there arduino standalo version?

ngraziano, 2024-10-02:
No arduino standalone version exist. I do not have time to maintain multiple version.
You can do it yourself the modification may not be hard.

areytechai, 2024-10-02:
i just tried so failed, can you simply assist me please?

 i need class -c support you have got very great solution of that,

so happy if you share me class c lpwa code

thanks

ngraziano, 2024-10-04:
Can you tell me more about your problem.

What is your platform ? (microcontroler)
What is the error ?
Why do you not want to try platformio ?

areytechai, 2024-10-04:
> Can you tell me more about your problem.
> 
> What is your platform ? (microcontroler) What is the error ? Why do you not want to try platformio ?

Thank you so much for response,

platform is ttgo esp32 oled, and heltec v3 but mostly i need for ttgoesp32 

2nd and 3rd questions are pretty foreign to platformio, so happy to assist a standalone class C code share with us,

thanks

ngraziano, 2024-10-04:
You can look at the exemple in examples/esp32_classC directory, this example is made for an heltec V1 board (LCD and chip on board), it connect to the network and show the string receive from the network on port 10.

You will have to set your keys in lorakeys.h and change the use of display in main.cpp : line 123 is where the message is receive, `showState`  is use to display the current state you may have to change it too.

If want to try platformio just follow instruction here https://docs.platformio.org/en/latest/integration/ide/vscode.html#ide-vscode and change the file platformio.ini to select your board ( for exemple https://docs.platformio.org/en/latest/boards/espressif32/heltec_wifi_lora_32_V3.html or https://docs.platformio.org/en/latest/boards/espressif32/ttgo-lora32-v21.html see https://registry.platformio.org/platforms/platformio/espressif32/boards)
and then compile the project, platformio will download all the needed dependencies and framework. 

If you want to use the code without platformio you will have to copy all the library in your project and and make sure the include path are good and change main.cpp to main.ino

areytechai, 2024-10-04:
Thank you so much, i tried but couldnt make it, probably needs library,

kindly can you share a zip file?

ngraziano, 2024-10-04:
Sorry don't have time for that

closed 2024-10-04 by ngraziano

areytechai, 2024-10-05:
oh why